sparkfun/SparkFun_Micro_OLED_Arduino_Library #2: Font size three text

DaAwesomeP, 2015-08-10:
When I use font size 3, I can only display numbers. Actual text leaves the display blank. I know that this is intended for font size 2 (the seven segment display), but is it intended for this font size too? It doesn't say that in the demo file nor the docs.

DaAwesomeP, 2015-08-11:
I was looking through the library source and I saw that the fonts seem to be HEX codes. That's definitely not something that I could do myself.

jimblom, 2015-08-11:
Yup, unfortunately the large and seven-segment fonts only have numbers defined. I'll make sure that's documented somewhere.

If you're interested in making your own fonts, [this tutorial](http://learn.microview.io/Font/creating-fonts-for-microview.html) might help (it's for the MicroView, but applies to this display as well).

closed 2015-08-11 by jimblom

DaAwesomeP, 2015-08-13:
Created a new font: https://github.com/sparkfun/SparkFun_Micro_OLED_Arduino_Library/pull/4