sparkfun/SparkFun_Micro_OLED_Arduino_Library #5: Text not printing with ESP8266 thing

emil01, 2015-08-17:
Hi,
As the title says, I am unable to get any text to print on the display when I am using the ESP8266 thing. 

As I was short on time, I used the adafruit_gfx library just for printing text and it worked straight out of the box. 

thanks
Emil Varughese

rangerchaz, 2016-03-14:
Hi,

I am having the same issue on a 8266 as well. @emil01, how were you able to get the adafruit_gfx library to work to print text? Did you use it with the SparkFun Micro OLED Arduino library? Or, did you use the adafruit_gfx library by itself?

Thanks,

Chad Cox

emil01, 2016-03-15:
@rangerchaz  Have a look at https://github.com/EdwinRobotics/SparkFun_Micro_OLED_Arduino_Library/tree/GSlibrary 

It is not a clean code but it works tested it right now.

ericcastro, 2016-04-04:
Hello, I have tested the EdwinRobotics fork and latest version from this repo but none can get the text to display correctly. I am using a WeMos D1 mini board that is based on the ESP8266, and their OLED shield that is based on the SSD1306 and has a 64x48 size.

I'm using this as I cannot get the Adafruit SSD1306 library to be compatible with the smaller screen size, also the u8glib library won't work on the ESP8266. So this is the library I got the best results with, I can display every example and do everything except drawing text.

Any ideas? this is what I get when I try to print the word "test" at offset 0, 10 with font type 1
http://i.imgur.com/9FQDruP.jpg

This is my code:
```arduino
#include <Wire.h>  // Include Wire if you're using I2C
#include <SFE_MicroOLED.h>  // Include the SFE_MicroOLED library
#define PIN_RESET D8  // Connect RST to pin 9 (req. for SPI and I2C)
#define DC_JUMPER D7

MicroOLED oled(PIN_RESET, DC_JUMPER); // Example I2C declaration

void setup()
{
  oled.begin();
  oled.clear(ALL);
  oled.display();   
  delay(1000);
  oled.clear(PAGE);
}

void loop()
{
  oled.setFontType(0);  // Set font to type 1
  oled.clear(PAGE);     // Clear the page
  oled.setCursor(0, 10); // Set cursor to top-left
  oled.print("test");
  oled.display();
  delay(1000);
}```

emil01, 2016-04-05:
@ericcastro I just tested https://github.com/EdwinRobotics/SparkFun_Micro_OLED_Arduino_Library/tree/GSlibrary on Sparkfun Thing board and text works fine

Are you able to get any output at all?

ericcastro, 2016-04-05:
Hi emil01 - did you test my code?

You can see the output I get on the image I uploaded on my previous post. I get gibberish when trying to draw text.

The cube and clock examples are working fine, so drawing is ok - but can't get text to work.

My board is a WeMos D1 mini also based on the ESP8266 and can only work with the OLED screen via I2C, whose SCL and SDA pins are in pins D2 and D1.

emil01, 2016-04-05:
Please find below your modified code which i used to get it working on my setup. Please refer to the sparkfun hookup guide about the DC_JUMPER settings. 

One bug that is there is the set cursor, which for some reason doesnt work as its supposed to always.

`#include <Wire.h>  // Include Wire if you're using I2C
#include <ER_MicroOLED.h>  // Include the SFE_MicroOLED library
#define PIN_RESET 12  // Connect RST to pin 9 (req. for SPI and I2C)
#define DC_JUMPER 1

MicroOLED oled(PIN_RESET, DC_JUMPER); // Example I2C declaration

void setup()
{
  oled.begin();
  oled.clear(ALL);
  oled.display();   
  delay(1000);
  oled.clear(PAGE);
}

void loop()
{
  oled.setFontType(0);  // Set font to type 1
  oled.clear(PAGE);     // Clear the page
  oled.setTextColor(WHITE);
  oled.setTextSize(0);  
  oled.setCursor(0, 10); // Set cursor to top-left
  oled.print("test");
  oled.display();
  delay(1000);
}`

ericcastro, 2016-04-05:
@emil01, the example doesn't work as you seem to use a different (possibly updated/modified) version the library called ER_MicroOLED, which I don't have; you are also using methods not present in the sparkfun's library such as setTextColor and setTextSize.

Also, I cannot choose the pins the OLED shield connects to because it's in a shield format and just by specifying which pins I2C is in should be enough (in my case, D2 and D1)

Can you provide your version of the library you are using, so I can try it out ?

emil01, 2016-04-05:
@ericcastro please refer to the link i had posted above, this is the library that i am using. I changed the pins as per my current setup thats all, you can use as you wish. 

https://github.com/EdwinRobotics/SparkFun_Micro_OLED_Arduino_Library/tree/GSlibrary

emil01, 2016-04-20:
@ericcastro @rangerchaz Fixed the setCursor not working part

closed 2016-04-20 by emil01

Sarah-C, 2016-09-02:
On line 45 and 46 in file SFE_MicroLED.cpp there is a fix for a warning in the compiler.

This fix BREAKS the storage of fonts in PROGMEM. (Hence the user up there fixing it by removing PROGMEM from the declaration)

To fix and keep the fonts in PROGMEM - just comment the two lines 45, and 46 out.

E.g:

44. // This fixed ugly GCC warning "only initialized variables can be placed into program memory area"
45. //#undef PROGMEM
46. //#define PROGMEM __attribute__((section(".progmem.data")))