sparkfun/SparkFun_Micro_OLED_Arduino_Library #22: Previous Definition?

tnsolley, 2019-02-13:
Arduino Nano
Arduino IDE 1.8.8
Windows 10

```
In file included from H:\Dev\Cosplay_pictures\Legacy_Mando_SWEU\Helmet\sketch_jan21a\sketch_jan21a.ino:56:0:

C:\Users\thomas.solley\Documents\Arduino\libraries\SFE_MicroOLED/SFE_MicroOLED.h:105:14: note: previous definition here

 typedef enum CMD {

              ^

In file included from H:\Dev\Cosplay_pictures\Legacy_Mando_SWEU\Helmet\sketch_jan21a\sketch_jan21a.ino:56:0:

C:\Users\thomas.solley\Documents\Arduino\libraries\SFE_MicroOLED/SFE_MicroOLED.h:125:3: note: previous declaration as 'typedef enum CMD commCommand_t'

 } commCommand_t;

   ^
```

oclyke, 2019-03-08:
@tnsolley Is this still an issue? I just compiled the "Hello" example for Nano in Arduino 1.8.8 on Windows10.

tnsolley, 2019-03-18:
@oclyke  Good question, and a fault on my part.
See issue 6 linked above; I had several libraries [4Duino Picasso Serial library, RTClib] which had `CMD` enumerated, by changing their names to a library-specific name the issue was resolved.

closed 2019-03-18 by tnsolley