sparkfun/SparkFun_Micro_OLED_Arduino_Library #24: Missing scrolling functions in .cpp

halveez, 2020-07-19:
I believe the cpp file is missing the definitions of the following scrolling functions:

	void scrollLeft(uint8_t start, uint8_t stop);
	void scrollVertRight(uint8_t start, uint8_t stop);
	void scrollVertLeft(uint8_t start, uint8_t stop);

scrollRight and scrollStop appear to be the only scrolling functions that are present in the cpp file.

halveez, 2020-07-19:
The reference link above the right scrolling functions appears to be dead as well:

http://learn.microview.io/intro/general-overview-of-microview.html

PaulZC, 2020-12-10:
Hi @halveez ,
Thanks for spotting this issue - and apologies for the slow reply!
We've added the missing functions in version 1.2.10 and - as a bonus - you can now set the scroll interval too.
Best wishes,
Paul

closed 2020-12-10 by PaulZC

halveez, 2020-12-11:
Thanks Paul!