sparkfun/SparkFun_Micro_OLED_Arduino_Library #26: isConnected test fails on unbranded micro OLED display

PaulZC, 2020-12-10:
Example6_DrawIcon uses the standard ```isConnected``` test to discover a display with address 0x3D or 0x3C.
The _**unbranded**_ OLED display I have fails this test:

[image]

If the SparkFun displays pass this test, then please close with extreme prejudice!
Paul

nseidle, 2020-12-10:
[Qwiic OLED](https://www.sparkfun.com/products/14532) happily acks at 0x3D (default address). 

[image]
Don't mind the 0x42 u-blox GNSS and 0x36 MAX17048 fuel gauge, I tested on the ESP32 [RTK Surveyor](https://www.sparkfun.com/products/17369) that has these built in.

What address does your unbranded display ack to?

PaulZC, 2020-12-11:
It's really very strange... It defaults to address 0x3C and ACKs OK when I write data to it:
[image]

But doesn't respond to ```isConnected``` either way round:
[image]

[image]

I suggest we don't lose any sleep over this as it is an _unbranded_ display, but maybe we should leave this issue open (and make it sticky?) just in case any users have the same issue?

PaulZC, 2020-12-11:
Thanks Pete - I'll give it a try!
All the best,
Paul

PaulZC, 2020-12-11:
So, pulling SDA low didn't seem to do the trick. But that got me thinking... And if I give the display a manual reset before trying ```isConnected``` then it does work! Cheers Pete - you're my inspiration! ;-D

@nseidle: should we update the example? I'm not seeing any downside?

[image]

nseidle, 2020-12-11:
Sounds good to me!

PaulZC, 2020-12-11:
Done. The [new version](https://github.com/sparkfun/SparkFun_Micro_OLED_Arduino_Library/commit/a36b432fd47c479646815657a4cd0dd3e9427643#diff-0cc0f1557ae9035f8ce4c65683e5abb861dfeaa7c575377c1c24df9daca24f85R83-R89) is in the release_candidate branch for now...
And that makes it beer-o'-clock!
Closing...

closed 2020-12-11 by PaulZC