sparkfun/SparkFun_Micro_OLED_Arduino_Library #40: RP2040/Pico port?

beshrkayali, 2021-09-01:
Hi,

Just wondering if there are any plans to provide a port for the new rp2040 / raspberry pi pico?

beshrkayali, 2021-09-01:
This issue was posted before RP2040 had Arduino support, so the Pico SDK was the only thing that could be meant.

And I have managed to port and use it my project at the time actually, I was just suggesting a way for you guys to provide additional value to your customers, but thank you for your casual condescension! You can keep this issue closed.

santaimpersonator, 2021-09-01:
@beshrkayali My apologies, if that came off as condescending. That wasn't my intention. I'm guessing this was the offensive statement:

> Additionally, if you intend to use the Pico SDK... you should have the skills/knowledge to port the library on your own.

We have a lot of customers, who unintentionally, get in over their heads when jumping into new and exciting products. I only meant to preface that customers who intend to use the Pico SDK, should be able to port the library on their own. This was to address if you were a more novice user, without any previous coding experience. It was also meant serve as a stopgap to keep novice users from "going down the rabbit hole" when we offer more technical support with the Arduino IDE/library. However, I will take your feedback and re-word that statement going forward with a longer, detailed explanation to avoid further misunderstandings.

Regarding porting the Arduino library to the Pico SDK:
* Currently, we don't intend to port over all of our Arduino libraries for the Pico SDK. We are currently focusing our efforts to port over the libraries for our I2C based products to Python for our [Qwiic Python package](https://github.com/sparkfun/Qwiic_Py).